openwrt/luci #4205: [luci-app-vpn-policy-routing] adding wildcard domain names to luci when using dnsmasq ipset

yazdan, 2020-06-25:
Hi,

First of all thank you all for your effort for providing me and others with a great product.

When using "vpn-policy-routing" package from cli, it is possible to add follwing configs

```
config policy
	option dest_addr '.youtube.com'
	option interface 'tunel'

config policy
	option interface 'tunel'
	option dest_addr '.youtu.be'

config policy
	option dest_addr '.ytimg.com'
	option interface 'tunel'

config policy
	option dest_addr '.ggpht.com'
	option interface 'tunel'

config policy                          
        option dest_addr '.googlevideo.com'     
        option interface 'tunel' 

config policy
	option interface 'tunel'

```
that is somehow treated as a wildcard domain and I have tested it.

But when using luci application, I get "invalid"
 when I try to save the values. 
[image]

My environment is as follow:
OpenWrt 19.07.3 r11063-85e04e9f46
TP-Link Archer C5 v1
LuCI openwrt-19.07 branch git-20.136.49537-fb2f363
vpn-policy-routing 0.2.1-13
luci-app-vpn-policy-routing git-20.167.61968-87da00a-73

I tried to find out where all validation happens and send a PR but could not find any clue. I'm new to openwrt specially luci development.

Thanks.

yazdan, 2020-06-25:
it seems when I specify source address this type of domain is not working

jow-, 2020-06-30:
@stangri

stangri, 2020-06-30:
@yazdan should work without the leading dot.

yazdan, 2020-06-30:
@stangri I need to route all sub-domains in wildcard fashion and when I add sites without leading dot it seems not working. But I try it and let you know. I may have mixed up things.

stangri, 2020-06-30:
With dnsmasq.ipset setting it should work for subdomains without the leading dot.

yazdan, 2020-06-30:
Aha Thanks a lot

jow-, 2020-07-04:
Closing this as it does not appear to be a bug.

closed 2020-07-04 by jow-