openwrt/luci #4210: Different channels in multiSSID setup for one wifi card

pali, 2020-06-26:
Some wifi cards support creating more AP interfaces on different channels. But luci currently does not support specifying different channel for different interfaces which belongs to the same card. Currently all interfaces which belongs to the same card (phy) share same channel/frequency.

iw phy info prints following output for such card:
```
        valid interface combinations:
                 * #{ managed, AP, P2P-client, P2P-GO } <= 3,
                   total <= 3, #channels <= 2, STA/AP BI must match
```

jow-, 2020-06-30:
This capability is not exposed by OpenWrt yet so it cannot be implemented in LuCI

closed 2020-06-30 by jow-

BKPepe, 2020-07-23:
For me, it sounds like an interesting feature that shouldn't be missing in OpenWrt! Because of that, I created feature request here:
https://bugs.openwrt.org/index.php?do=details&task_id=3243

Hopefully, it won't be `/dev/null` among those many "bugs" which are already there. In the meantime, there could be documentation that would supply it until it is implemented in OpenWrt and if LuCI would be prepared for that, it will be amazing!

Neustradamus, 2020-08-01:
@jow-: Can you reopen this ticket?

jow-, 2020-08-01:
I’d rather avoid having yet another eternal feature request ticket hanging around. If the basics are implemented in OpenWrt, LuCI will follow accordingly.

In order to implement it in OpenWrt in the first place, this ticket is the wrong spot