openwrt/luci #4216: luci-app-cshark: Ca validated by default, but cloudshark.org's ca is not trusted

mikeasolo, 2020-06-30:
Problem Description:
	Cshark fails silently in LUCI
	Cshark fails with "invalid SSL certificate". 
	The included certificate is expired and a different CA than used by cloudshark.org
	Manual updating of Cloudshark CA trust file increases the barrier to entry. 

Reproduction of issue
	root@OpenWrt:~# cshark
	capturing traffic to file: '/tmp/cshark.pcap-pGcEPi' ...
	^C
	420 packets captured
	uploading capture ...
	cshark[5010]: cshark: invalid SSL certificate
	cshark: invalid SSL certificate

Fix options:
	Expose CA_Verify in Luci as an option and explaining the trusted certificate path
	Default CA_Verify to false
	Update/add lets encrypt cert for www.cloudshark.org
	
cat /etc/config/cshark
config cshark 'cshark'
        option url 'https://www.cloudshark.org' # uses Let's Encrypt Authority X3 
        option ca '/etc/ssl/certs/ca-the_usertrust_network.pem'
        option ca_verify '1' #set to 0 to work around this issue
        option dir '/tmp/'
        option token '<MyAPiKey>'

cat /etc/ssl/certs/ca-the_usertrust_network.pem' #Old certificate authority
	Common Name: AddTrust External CA Root
	Organization: AddTrust AB
	Organization Unit: AddTrust External TTP Network
	Country: SE
	Valid From: May 30, 2000
	Valid To: May 30, 2020
	Issuer: AddTrust External CA Root, AddTrust AB
	Serial Number: 1 (0x1)

	-----BEGIN CERTIFICATE-----
	MIIENjCCAx6gAwIBAgIBATANBgkqhkiG9w0BAQUFADBvMQswCQYDVQQGEwJTRTEU
	MBIGA1UEChMLQWRkVHJ1c3QgQUIxJjAkBgNVBAsTHUFkZFRydXN0IEV4dGVybmFs
	IFRUUCBOZXR3b3JrMSIwIAYDVQQDExlBZGRUcnVzdCBFeHRlcm5hbCBDQSBSb290
	MB4XDTAwMDUzMDEwNDgzOFoXDTIwMDUzMDEwNDgzOFowbzELMAkGA1UEBhMCU0Ux
	FDASBgNVBAoTC0FkZFRydXN0IEFCMSYwJAYDVQQLEx1BZGRUcnVzdCBFeHRlcm5h
	bCBUVFAgTmV0d29yazEiMCAGA1UEAxMZQWRkVHJ1c3QgRXh0ZXJuYWwgQ0EgUm9v
	dDCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBALf3GjPm8gAELTngTlvt
	H7xsD821+iO2zt6bETOXpClMfZOfvUq8k+0DGuOPz+VtUFrWlymUWoCwSXrbLpX9
	uMq/NzgtHj6RQa1wVsfwTz/oMp50ysiQVOnGXw94nZpAPA6sYapeFI+eh6FqUNzX
	mk6vBbOmcZSccbNQYArHE504B4YCqOmoaSYYkKtMsE8jqzpPhNjfzp/haW+710LX
	a0Tkx63ubUFfclpxCDezeWWkWaCUN/cALw3CknLa0Dhy2xSoRcRdKn23tNbE7qzN
	E0S3ySvdQwAl+mG5aWpYIxG3pzOPVnVZ9c0p10a3CitlttNCbxWyuHv77+ldU9U0
	WicCAwEAAaOB3DCB2TAdBgNVHQ4EFgQUrb2YejS0Jvf6xCZU7wO94CTLVBowCwYD
	VR0PBAQDAgEGMA8GA1UdEwEB/wQFMAMBAf8wgZkGA1UdIwSBkTCBjoAUrb2YejS0
	Jvf6xCZU7wO94CTLVBqhc6RxMG8xCzAJBgNVBAYTAlNFMRQwEgYDVQQKEwtBZGRU
	cnVzdCBBQjEmMCQGA1UECxMdQWRkVHJ1c3QgRXh0ZXJuYWwgVFRQIE5ldHdvcmsx
	IjAgBgNVBAMTGUFkZFRydXN0IEV4dGVybmFsIENBIFJvb3SCAQEwDQYJKoZIhvcN
	AQEFBQADggEBALCb4IUlwtYj4g+WBpKdQZic2YR5gdkeWxQHIzZlj7DYd7usQWxH
	YINRsPkyPef89iYTx4AWpb9a/IfPeHmJIZriTAcKhjW88t5RxNKWt9x+Tu5w/Rw5
	6wwCURQtjr0W4MHfRnXnJK3s9EK0hZNwEGe6nQY1ShjTK3rMUUKhemPR5ruhxSvC
	Nr4TDea9Y355e6cJDUCrat2PisP29owaQgVR1EX1n6diIWgVIEM8med8vSTYqZEX
	c4g/VhsxOBi0cQ+azcgOno4uG+GMmIPLHzHxREzGBHNJdmAPx/i9F4BrLunMTA5a
	mnkPIAou1Z5jJh5VkpTYghdae9C8x49OhgQ=
	-----END CERTIFICATE-----

Lets Encrypt  #new certificate authority
	Common Name: Let's Encrypt Authority X3
	Organization: Let's Encrypt
	Country: US
	Valid From: March 17, 2016
	Valid To: March 17, 2021
	Issuer: DST Root CA X3, Digital Signature Trust Co.
	Serial Number: 0a0141420000015385736a0b85eca708

	-----BEGIN CERTIFICATE-----
	MIIEkjCCA3qgAwIBAgIQCgFBQgAAAVOFc2oLheynCDANBgkqhkiG9w0BAQsFADA/
	MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT
	DkRTVCBSb290IENBIFgzMB4XDTE2MDMxNzE2NDA0NloXDTIxMDMxNzE2NDA0Nlow
	SjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxIzAhBgNVBAMT
	GkxldCdzIEVuY3J5cHQgQXV0aG9yaXR5IFgzMIIBIjANBgkqhkiG9w0BAQEFAAOC
	AQ8AMIIBCgKCAQEAnNMM8FrlLke3cl03g7NoYzDq1zUmGSXhvb418XCSL7e4S0EF
	q6meNQhY7LEqxGiHC6PjdeTm86dicbp5gWAf15Gan/PQeGdxyGkOlZHP/uaZ6WA8
	SMx+yk13EiSdRxta67nsHjcAHJyse6cF6s5K671B5TaYucv9bTyWaN8jKkKQDIZ0
	Z8h/pZq4UmEUEz9l6YKHy9v6Dlb2honzhT+Xhq+w3Brvaw2VFn3EK6BlspkENnWA
	a6xK8xuQSXgvopZPKiAlKQTGdMDQMc2PMTiVFrqoM7hD8bEfwzB/onkxEz0tNvjj
	/PIzark5McWvxI0NHWQWM6r6hCm21AvA2H3DkwIDAQABo4IBfTCCAXkwEgYDVR0T
	AQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8EBAMCAYYwfwYIKwYBBQUHAQEEczBxMDIG
	CCsGAQUFBzABhiZodHRwOi8vaXNyZy50cnVzdGlkLm9jc3AuaWRlbnRydXN0LmNv
	bTA7BggrBgEFBQcwAoYvaHR0cDovL2FwcHMuaWRlbnRydXN0LmNvbS9yb290cy9k
	c3Ryb290Y2F4My5wN2MwHwYDVR0jBBgwFoAUxKexpHsscfrb4UuQdf/EFWCFiRAw
	VAYDVR0gBE0wSzAIBgZngQwBAgEwPwYLKwYBBAGC3xMBAQEwMDAuBggrBgEFBQcC
	ARYiaHR0cDovL2Nwcy5yb290LXgxLmxldHNlbmNyeXB0Lm9yZzA8BgNVHR8ENTAz
	MDGgL6AthitodHRwOi8vY3JsLmlkZW50cnVzdC5jb20vRFNUUk9PVENBWDNDUkwu
	Y3JsMB0GA1UdDgQWBBSoSmpjBH3duubRObemRWXv86jsoTANBgkqhkiG9w0BAQsF
	AAOCAQEA3TPXEfNjWDjdGBX7CVW+dla5cEilaUcne8IkCJLxWh9KEik3JHRRHGJo
	uM2VcGfl96S8TihRzZvoroed6ti6WqEBmtzw3Wodatg+VyOeph4EYpr/1wXKtx8/
	wApIvJSwtmVi4MFU5aMqrSDE6ea73Mj2tcMyo5jMd6jmeWUHK8so/joWUoHOUgwu
	X4Po1QYz+3dszkDqMp4fklxBwXRsW10KXzPMTZ+sOPAveyxindmjkW8lGy+QsRlG
	PfZ+G6Z6h7mjem0Y+iWlkYcV4PIWL1iwBi8saCbGS5jN2p8M+X+Q7UNKEkROb3N6
	KOqkqm57TH2H3eDJAkSnh6/DNFu0Qg==
	-----END CERTIFICATE-----

version information:
	root@OpenWrt:/etc# cshark -v
	cshark version v0.1
	root@OpenWrt:/etc#  opkg list | grep cshark
	cshark - 2018-08-20-7a7cf7f3-1
	luci-app-cshark - git-20.172.67167-b397549
	OpenWrt SNAPSHOT r13601-d93da0d016

jow-, 2020-06-30:
@danrl

danrl, 2020-06-30:
Thanks for the detailed and well-written issue report @mikeasolo.

IIRC the certificates cshark uses are the ones installed system-wide. Am I mistaken here maybe?
Isn't Lets Encrypt's "new" root cert part of the system-wide trust store? I'm a bit puzzled.

(It's been a while for me since I last looked into cshark, let alone in luci)

danrl, 2020-06-30:
My bad, of course cshark was using it's own CA file (usertrust network).

I don't have a platform to fix and test on, but here's the recipe for whoever likes to give it a shot:

1. Fix upstream: [https://github.com/cloudshark/cshark/tree/master/config](config files) to point to '/etc/ssl/certs/ca-certificates.crt' (OpenWrt system wide CA certs)
2. Update *package* cshark [Makefile](https://github.com/openwrt/packages/blob/78248fb2fdc34ea96c8bd12e0dd470b340ed0d1d/net/cshark/Makefile) to point to fixed upstream version. Consider removing the usertrust network references to keep the house clean.
3. Profit :)

mikeasolo, 2020-06-30:
> My bad, of course cshark was using it's own CA file (usertrust network).
> I don't have a platform to fix and test on, but here's the recipe for whoever likes to give it a shot:
I'll give it a shot