openwrt/luci #4217: luci-app-commands: broken special characters

koshev-msk, 2020-06-30:
Hi! I am using luci application commands.
I have to push the command and argument with special characters.
Input Example: mycommand AT+QCSQ
and see in the output answer
"mycommand AT QCSQ"
The next time I enter the argument: AT%2BQCSQ and I see a special character "+" in the response.
Possible fix it?
Thanks.

jow-, 2020-06-30:
Which LuCI version?

koshev-msk, 2020-06-30:
Tested on 
luci git-20.064.26341-4fc4c79 OpenWrt 18.06.8
and 
luci git-20.178.63701-ce35d08 OpenWrt 19.07.3

I seen same problem.

jow-, 2020-07-04:
Cannot reproduce on either 19.07 or master.

19.07:
[image]
[image]

master:
[image]
[image]

closed 2020-07-04 by jow-