openwrt/luci #4226: luci-app-smartdns RPCError

snakwu, 2020-07-03:
RPCError
RPC call to uci/get failed with ubus code 6: 没有权限
  at ClassConstructor.handleCallReply (https://192.168.88.1/luci-static/resources/rpc.js?v=git-20.182.56619-8f94331:16:3)

feckert, 2020-07-03:
@snakwu Please add read/write permissions to https://github.com/openwrt/luci/blob/master/applications/luci-app-smartdns/root/usr/share/rpcd/acl.d/luci-app-smartdns.json

Please have a look for uci read/write permissions howto solve this to the luci-app-adblock.
Write:
https://github.com/openwrt/luci/blob/master/applications/luci-app-adblock/root/usr/share/rpcd/acl.d/luci-app-adblock.json#L5
Read:
https://github.com/openwrt/luci/blob/master/applications/luci-app-adblock/root/usr/share/rpcd/acl.d/luci-app-adblock.json#L28

zxlhhyccc, 2020-07-07:
acl文件改成以下格式正常：
{
	"luci-app-smartdns": {
		"description": "Grant access to LuCI app smartdns",
		"read": {
			"file": {
				"/etc/smartdns/*": [ "read" ],
				"/usr/sbin/iptables -t nat -nL PREROUTING": [ "exec" ],
				"/usr/sbin/ip6tables -t nat -nL PREROUTING": [ "exec" ],
				"/usr/sbin/smartdns": [ "exec" ]
			},
			"ubus": {
				"service": [ "list" ]
			},
			"uci": [ "smartdns" ]
		},
		"write": {
			"file": {
				"/etc/smartdns/*": [ "write" ]
			},
			"uci": [ "smartdns" ]
		}
	}
}

feckert, 2020-07-07:
Thanks
Fixed with https://github.com/openwrt/luci/commit/1b30e20eaa0d0db6686697f44764168edd11b757

commit 1b30e20e (2020-07-07): luci-app-smartdns: fix acl for uci

fixes #4226
closed 2020-07-07 by feckert