openwrt/luci #4238: LuCI does not recognize tcpudp value in firewall

BKPepe, 2020-07-07:
While looking into Firewall configuration (file: /etc/config/firewall), I found in [the documentation](https://openwrt.org/docs/guide-user/firewall/firewall_configuration), there is a default value `tcpudp` for `proto` and as well in the description. In firewall3 repository, I found that here https://git.openwrt.org/?p=project/firewall3.git;a=blob;f=rules.c;h=181c6b1a126f4e4f1f7f23f5620bb20d1bd08c6a;hb=HEAD#l209 and here https://git.openwrt.org/?p=project/firewall3.git;a=blob;f=options.c;h=aed0cfb8a441dcd44ca9c478add57aa49bed20cd;hb=refs/heads/master#l608 is mentioned protocol tcpudp.

In LuCI, I can not set it and if I change it under the hood by using it manually, I endup with unrecognized protocol.
[image]

What's the correct solution? Using something like this
```
        list proto 'tcp'
        list proto 'udp'
```

or this one according to the docs?
```
        option proto 'tcpudp'
```

This happens on the latest OpenWrt 19.07.3, I didn't try OpenWrt master so far.

jow-, 2020-07-07:
Use the former notation. Using `option proto 'tcp udp'` should work too.

adelton, 2025-01-11:
Grepping the code I came across
https://github.com/openwrt/luci/blob/master/applications/luci-app-firewall/htdocs/luci-static/resources/tools/firewall.js#L554-L557
which looks like an attempt to turn `tcpudp` value from the config file into the two selections. However, at least on OpenWrt 24.10.0-rc5 (r28304-6dacba30a7) that does not have the desired effect, I still see the verbatim `tcpudp` and Unrecognized protocol message in the UI when I put
```
        option proto 'tcpudp' 
```
manually to a rule in `/etc/config/firewall`.

Would someone know if this is the place that was expected to take care of that `tcpudp` value? Perhaps @jow- who added that logic in 7944b0a90bbb0f0f5d1da5eaf5d9906ac546c44b?