openwrt/luci #4247: luci-app-ttyd not show in services menu at all in the web page.

huyu9711, 2020-07-09:
After install it, not menu item in services menu at all.

atiensivu, 2020-07-12:
It shows up for me on a recent snapshot build. I have Luci configured not to use SSL.

In my case, ttyd never renders inside the browser, but if I look at the debugging console, I notice a warning that seems to be coming from this page:

luci/applications/luci-app-ttyd/htdocs/luci-static/resources/view/ttyd/term.js

Specifically:
[blocked] The page at about:blank was not allowed to display insecure content from http://192.168.9.1:7681/.

That is using Safari Technology Preview Release 109 (Safari 14.0, WebKit 16610.1.17.2)
Snapshot build: LuCI Master (git-20.172.67167-b397549) / OpenWrt SNAPSHOT r13611-3f27a6e640

If I click on the highlighted "http://192.168.9.1:7681/", I do launch the ttyd window successfully.
Obviously, my instance is listening on 192.168.9.1

This may be better as its own issue, now that I think about it.

jow-, 2020-07-17:
As pointed out in the release notes for 19.07.3, a reboot is required for it to show in the menu. This is a regression which will be fixed with the next service release.

closed 2020-07-17 by jow-