openwrt/luci #4253: RPC error occurs when luci-app-smartDNS needs to be saved and applied when starting or closing the master branch

zxlhhyccc, 2020-07-10:
[image]

pymumu, 2020-07-12:
Please describe the openwrt version and the steps in which the problem occurred.

zxlhhyccc, 2020-07-17:
> Please describe the openwrt version and the steps in which the problem occurred.

Under the official master branch, compile JS Luci-app-SmartDNS. After adding upstream DNS or setting TCP server and other options, click "Save and apply", the above problems will occur. After cancellation, "save and apply" will work normally.

zxlhhyccc, 2020-07-17:
> Please describe the openwrt version and the steps in which the problem occurred.

I raised the issue in your warehouse！

adelton, 2025-01-11:
With OpenWrt 24.10.0-rc5 (r28304-6dacba30a7) and luci-app-smartdns 1.2023.42-r1 I cannot reproduce the issue. I'm able to add an Upstream Server and `Save & Apply`.