openwrt/luci #4260: Luci Statistics Disk Space Usage Missing Fields

lhartmann, 2020-07-12:
**PROBLEM:** From [collectd df plugin page](https://www.collectd.org/wiki/index.php/Plugin:DF) we get that it is capable of reporting space in both bytes and percentage, but these options are missing on `LUCI` / `Statistics` / `Setup` / `General Plugins` / `Disk Space Usage` / `Additional Field`.

**REQUEST:** Please add boolean fields for `ValuesAbsolute` and `ValuesPercentage`.

**USE CASE:** I am collecting to InfluxDB which stores free, used and reserved spaces in different rows of the database, and AFAIK it can not yet perform math across rows. I am plottig with Grafana, which also can't combine multiple separate queries. The problem is solved with collectd reporting percentage, but LUCI got in the way.

feckert, 2020-07-13:
Could you please check this pullrequest if this works for you?
https://github.com/openwrt/luci/pull/4267

I must mention that I did not do a test, because I do not use the module.
But from the best of my knowledge and belief I would say that this could solve your problem.

feckert, 2020-07-20:
@lhartmann does this [P/R](https://github.com/openwrt/luci/pull/4267) fixes your issue?

closed 2020-07-22 by feckert