openwrt/luci #4271: missing luci-app-nut from  luci

bjlockie, 2020-07-16:
There is no menu for luci-app-nut on a snapshot build and a 19.* build.
See this thread for more info:
[https://forum.openwrt.org/t/where-is-luci-app-nut-supposed-to-be/68929/5](url)

bobafetthotmail, 2020-07-17:
another app has a similar issue https://github.com/openwrt/luci/issues/4247

jow-, 2020-07-17:
You need to install `nut-server` and reboot for the application to show up.

bobafetthotmail, 2020-07-17:
shouldn't nut-server be a dependency then?

jow-, 2020-07-17:
No idea, I didn't write the app. But judging from how the controller is written, all parts are optional and it will only show up if one of the optional components is present:
https://github.com/openwrt/luci/blob/openwrt-19.07/applications/luci-app-nut/luasrc/controller/nut.lua

I only figured out that `nut-server` provides `/etc/config/nut_server`. Not sure which packages provide these other configs.

bobafetthotmail, 2020-07-17:
I don't see any reasons against splitting it up in 3 luci-app packages (server/monitor/cgi) so each luci interface app can have its own nut backend dependency, and avoid confusing users again.

Seems easy enough, I can do that. Would that be ok?

jow-, 2020-07-17:
Not sure if it makes sense to make it excessively modular. Maybe it should just be ported to client side and made dependent on nut-server. I guess this is the most common use case. I'd appreciate if you'd look into it.

bjlockie, 2020-07-17:
I have all those packages installed via imagebuilder:

make image PROFILE=rpi-4 PACKAGES="kmod-usb-net-rtl8152 kmod-mt76 kmod-mt76x2u kmod-mt76x02-usb kmod-mt76x2 kmod-mt76-usb kmod-mt76x2-common kmod-mt76x2u luci usbutils wireless-tools pciutils luci-app-nlbwmon sqlite3-cli bcm27xx-userland openssl-util htop wireguard wireguard-tools kmod-wireguard luci-app-wireguard **nut-upsc nut-server nut-driver-usbhid-ups nut-upsmon luci-app-nut** vpn-policy-routing luci-app-vpn-policy-routing python3 sysfsutils -cypress-firmware-43455-sdio -cypress-firmware-43455-sdio-rpi-4b -brcmfmac-firmware-usb -kmod-brcmfmac -kmod-brcmutil" FILES=files/`

jow-, 2020-07-18:
*shrug* I installed luci-app-nut and nut-server on a 19.07.3 x86 /64 image and rebooted, nut was present in the services menu.

bjlockie, 2020-07-19:
I upgraded to the latest snapshot and the Services menu is there. :-)

bjlockie, 2020-07-19:
Oops.
The menu is there but it doesn't work:
`/usr/lib/lua/luci/dispatcher.lua:922: attempt to index local 'page' (a nil value)
stack traceback:
	/usr/lib/lua/luci/dispatcher.lua:922: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:984: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>`

fseifert77, 2021-01-26:
Issue closed? Problem still exists with latest snapshot from today. So what's the solution?
/usr/lib/lua/luci/dispatcher.lua:923: attempt to index local 'page' (a nil value)
stack traceback:
	/usr/lib/lua/luci/dispatcher.lua:923: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:985: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

xhemp, 2021-06-07:
Still happens on 21.02rc1.

OUZYcn, 2021-11-27:
same issue on  OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-21.231.26241-422c175

one more thing is ups driver **snmp-ups** can't find by opkg but should be there according to https://openwrt.org/docs/guide-user/services/ups/software.nut

OUZYcn, 2021-11-30:
> 
SO i tried compiled from source code for my ERX box. SNMP-UPS driver is could be select from menuconfig. 

[image]

[image]

closed 2025-02-27 by systemcrash