openwrt/luci #4282: wireguard MTU cannot exceed 1420

zhangyoufu, 2020-07-22:
https://github.com/openwrt/luci/blob/b8ff46c6d6f403b5881f812d91611b6b5b2167ed/protocols/luci-proto-wireguard/htdocs/luci-static/resources/protocol/wireguard.js#L77-L79

```
ethernet_mtu = 1500
sizeof_ip4_hdr = 20
sizeof_tcp_hdr = 20
sizeof_udp_hdr = 8
wireguard_overhead = 32
ip4_wireguard_mtu = ethernet_mtu - sizeof_ip4_hdr - sizeof_udp_hdr - wireguard_overhead = 1440
ip4_wireguard_ip4_tcp_mss = ip4_wireguard_mtu - sizeof_ip4_hdr - sizeof_tcp_hdr = 1400
```

Per my understanding, a MTU of 1440 should be acceptable.

zhangyoufu, 2020-07-22:
I have tested MTU 1440 with `ping -s 1398` (inetutils-ping). IPv4 header + payload inside wireguard tunnel = 1426 bytes.

zhangyoufu, 2020-07-22:
I have tested MTU 1440 in my setup. This MTU value maximize link utilization.

yousong, 2020-07-22:
1440 should be fine for ipv4 only traffic.

I think it's also worth noting that mtu of wireguard interfaces should be multiples of 16 as the protocol will pad payload up to for crypto operation.  This is especially the case if endpoint mtus do not match and both set their wg mtu values to the extremes (subtract out 60 without rounding down).

zhangyoufu, 2021-03-08:
> I think it's also worth noting that mtu of wireguard interfaces should be multiples of 16 as the protocol will pad payload up to for crypto operation. This is especially the case if endpoint mtus do not match and both set their wg mtu values to the extremes (subtract out 60 without rounding down).

@yousong I found your message at https://lists.zx2c4.com/pipermail/wireguard/2019-July/004289.html.

---

Although [Wireguard protocol doc](https://www.wireguard.com/protocol/#subsequent-messages-exchange-of-data-packets)  states
> encapsulated_packet = encapsulated_packet || zero padding in order to make the length a multiple of 16

The implementation does take care of MTU
```
static unsigned int calculate_skb_padding(struct sk_buff *skb)
{
	unsigned int padded_size, last_unit = skb->len;
	...
	padded_size = min(PACKET_CB(skb)->mtu,
			  ALIGN(last_unit, MESSAGE_PADDING_MULTIPLE));
	return padded_size - last_unit;
}
```

I set up a peer-to-peer test environment, wireguard over IPv6 over Ethernet is capable of transmitting MTU=1420 (1500-40-8-32) without fragmentation, even though 1420 is not a multiple of 16.

yousong, 2021-03-09:
> I set up a peer-to-peer test environment, wireguard over IPv6 over Ethernet is capable of transmitting MTU=1420 (1500-40-8-32) without fragmentation, even though 1420 is not a multiple of 16.

Reading the code again, I think you are right.  The code does say so and it works.  Sad that I do not recall the exact configuration at that time when the frags were observed with tcpdump and resolved when mtu was rounded down.

feckert, 2022-04-12:
Fixed with commit https://github.com/openwrt/luci/commit/ddd04a719f3adf4c59d6ab153de0c26eb9495a93

closed 2022-04-12 by feckert