openwrt/luci #4290: luci-app-adblock: allowlist/denylist strip underscores when saving

DHowett, 2020-07-26:
When editing an allowlist/denylist that contain underscores, the underscores are removed on save.

https://github.com/openwrt/luci/blob/master/applications/luci-app-adblock/htdocs/luci-static/resources/view/adblock/whitelist.js#L11

https://github.com/openwrt/luci/blob/master/applications/luci-app-adblock/htdocs/luci-static/resources/view/adblock/blacklist.js#L11

The regexes here exclude `_`.

dibdot, 2020-07-26:
What's your issue actually? Underscores  are not allowed in domain names.

DHowett, 2020-07-26:
So, I was attempting to allowlist `reddit_webonly.app.link`:

```

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;reddit_webonly.app.link.       IN      A

;; ANSWER SECTION:
reddit_webonly.app.link. 59     IN      A       143.204.25.10
reddit_webonly.app.link. 59     IN      A       143.204.25.126
reddit_webonly.app.link. 59     IN      A       143.204.25.4
reddit_webonly.app.link. 59     IN      A       143.204.25.68

;; Query time: 42 msec
;; SERVER: 1.1.1.1#53(1.1.1.1)
;; WHEN: Sat Jul 25 21:38:52 PDT 2020
;; MSG SIZE  rcvd: 139
```

I'm surprised to learn of a working, if not widespread, violation of RFC1035... but not terribly so. 😄

dibdot, 2020-07-26:
Please use hyphens instead ... or edit black-/whitelist only in a ssh session. :smile: 
I'm not going to change the current behavior.

closed 2020-07-26 by dibdot