openwrt/luci #4293: luci-app-adblock: TypeError result[0].trim is not a function

albilu, 2020-07-26:
Environment: 
Name	OpenWrt
Modele	TP-Link Archer C7 v5
Architecture	Qualcomm Atheros QCA956X ver 1 rev 0
Firmware Version	OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.204.30585-8750495
Kernel Version 	4.14.180

Description:
Adblock service running but unable to get overview in LuCi:
[image]

I already check related issues:
https://github.com/openwrt/packages/issues/11879
https://github.com/openwrt/luci/issues/3931
https://github.com/openwrt/luci/issues/3894

And none of the solutions provided there works.

Here is my packages versions:
`opkg list-installed | grep adblock`
adblock - 4.0.6-1
luci-app-adblock - git-20.205.21700-e8cba39
luci-i18n-adblock-fr - git-20.204.30585-8750495-1

`opkg list-installed | grep rpcd`
rpcd - 2020-05-26-67c8a3fd-1
rpcd-mod-file - 2020-05-26-67c8a3fd-1
rpcd-mod-iwinfo - 2020-05-26-67c8a3fd-1
rpcd-mod-luci - 20191114
rpcd-mod-rpcsys - 2020-05-26-67c8a3fd-1
rpcd-mod-rrdns - 20170710

and `opkg list-upgradable` is returning nothing so i have nothing left to upgrade

I was using 18.06.2 and everything was working fine. Please give details about the solutions to this i am regretting upgrading to 19.07.3 right now

dibdot, 2020-07-27:
Are you really serious? A series of tickets for client side issues in your setup? Reset all your Firefox browser caches or better use another browser like Chrome for testing. After that, close the other tickets as well ...

closed 2020-07-27 by dibdot

albilu, 2020-07-27:
What do you mean by client side issue I already clear browser caches, I used Firfoxe and chrome even chrome on Android phone.
Got the same error.

Please you can not just close the record without giving a proper solution.

Titcher, 2022-05-24:
For those still suffering from this bug while doing all relevant cache clears. This issue may be relevant: #5501