openwrt/luci #4297: luci-app-base: No related RPC reply

albilu, 2020-07-26:
Environment:
Name OpenWrt
Modele TP-Link Archer C7 v5
Architecture Qualcomm Atheros QCA956X ver 1 rev 0
Firmware Version OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.204.30585-8750495
Kernel Version 4.14.180

Description:
No related RPC reply
[image]

jow-, 2020-07-27:
Seems to be a client specific issue, in particular a flaky connection. Make sure your network is functioning properly, that you do not have any IP subnet conflicts, your connection is stable etc.

Also rule out any interfering browser addons, firewalls etc.

closed 2020-07-27 by jow-

w3spi5, 2021-11-23:
Same issue for me, when I'm using OpenVPN.
WHen arrive at home, admin connection is impossible and WIFI connection to WAN too