openwrt/luci #4298: luci-app-sysupgrade: XHR request timed out => i Can not upload firmware file

albilu, 2020-07-26:
Environment:
Name OpenWrt
Modele TP-Link Archer C7 v5
Architecture Qualcomm Atheros QCA956X ver 1 rev 0
Firmware Version OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.204.30585-8750495
Kernel Version 4.14.180

Description:
Unable to load any firmware file getting XHR request timed out 
[image]

jow-, 2020-07-27:
Seems to be a client specific issue, in particular a flaky connection. Make sure your network is functioning properly, that you do not have any IP subnet conflicts, your connection is stable etc.

Also rule out any interfering browser addons, firewalls etc.

closed 2020-07-27 by jow-

erdoukki, 2022-04-12:
> Seems to be a client specific issue
I can confirm that with same behavior on 21.02.2 was resolved by restarting the browser !