openwrt/luci #4300: Wrong Openwrt version in LUCI

epiniguin, 2020-07-27:
Hi!

I build 19.07.3 firmware for my D-Link DIR-835 using Image Builder.

In SSH I see right version:
```
root@dir835:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="19.07.3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.3"
VERSION_ID="19.07.3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r11063-85e04e9f46"
OPENWRT_BOARD="ar71xx/generic"
OPENWRT_ARCH="mips_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.3 r11063-85e04e9f46"

```

But in LUCI the version is 19.07.2:
[image]

jow-, 2020-07-27:
Ensure that your `/etc/openwrt_release` file is properly updated. Maybe it was accidentally restored form a backup or included in a `FILES` overlay.

epiniguin, 2020-07-27:
@jow- you are right!
That files were from a backup.

closed 2020-07-27 by epiniguin