openwrt/luci #4303: rpcd-mod-luci: only reports active leases from one dnsmasq server

aaronjg, 2020-07-28:
If multiple dnsmasq servers are running, only the first lease file is opened by `lease_open` in `libs/rpcd-mod-luci/src/luci.c` will only open the first one that it finds, rather than opening all the lease files from all instances of dnsmasq to find all the dhcp leases that have been issued.

socketbox, 2020-08-12:
Is this related to https://github.com/openwrt/luci/issues/911 ?

aaronjg, 2020-08-12:
Wow. This is a really old issue. Yes, this issue is a duplicate of #911. It looks like it was never patched. 

How does P/R #4308 look to fix this?

socketbox, 2020-08-13:
> How does P/R #4308 look to fix this?

Sorry, I haven't a clue. You might consider linking to the PR from that issue, or communicating with one of the three users in the associated thread.

jow-, 2020-11-06:
Fixed via 0c7a33523

closed 2020-11-06 by jow-