openwrt/luci #4321: luci-app-dockerman: Console Panel failure

zdz2019, 2020-07-31:
[image]

The command window cannot be displayed normally when the connect button is clicked.
No problem with the previous version.

@feckert

feckert, 2020-08-03:
I am not quite happy how the ttyd of a docker container is configured and started here.
But for now the [fix](https://github.com/TDT-AG/luci/commit/0b8e768765ad0d6351e0363f208f80dcf17cc128) here should help to get it working again without ssl.

Please let me know if this fix your issue.
I have mention that this only works if you did not redirect any http to https in uhttpd

I'm working on how to do this properly via ssl and so secure connection.

zdz2019, 2020-08-03:
I just tested your suggestion and it doesn't work, The page is always loading.

feckert, 2020-08-03:
@zdz2019 did you real tested by last push version of this commit?
If this is what I hear, it could be that you have tested a previous version of the fix, where I forgot that `&` while executing the ttyd command.

zdz2019, 2020-08-03:
I tested your commit 0b8e768765ad0d6351e0363f208f80dcf17cc128

feckert, 2020-08-04:
I have update my P/R in the mean time of your test and that was not the final change.
Please try https://github.com/TDT-AG/luci/commit/c851bd2645e32eb3b7146069d5f9012f71b406a4

zdz2019, 2020-08-04:
I test again,it worked well

PR #4332 (2020-08-11): luci-app-dockerman: improvements
- Clean up long source code lines that are hard to maintain
- Reorder menu entry so that the overview is the first entry
- Fixes socket definition with additional slash
- Fixes console connection (fixes #4321)
closed 2020-08-04 by feckert