openwrt/luci #4323: GUI issue after sysupgrade from 15.05.1 snapshot to 18.06.8.

roshavagarga, 2020-08-01:
The device in question is a WR841N v9. I was running an ancient snapshot somebody had made for me years ago and decided to do a sysupgrade to the newest tiny version with 'keep settings' checked.

During every restart I get this:
```
/usr/lib/lua/luci/dispatcher.lua:234: /etc/config/luci seems to be corrupt, unable to find section 'main'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:234: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:127: in function </usr/lib/lua/luci/dispatcher.lua:126>
```

Now, I know this might be an issue with `rpcd` not running, but I am what is known as an utter noob. Running Windows 7, putty and winscp both gave me 'connection refused' issues when I tried to use those to connect to the router and even when I do somehow manage that, I'd be grateful if somebody could point me to a ELI5 tutorial on how to get rpcd running properly on every restart.

I know the router is old, I'm basically using it for a 2.4ghz router, while my main router is only running 5ghz. I will note that despite the GUI not running, the wi-fi is running properly, just wanted to make sure I can fix it in case I want to tweak any settings.

translucentfocus, 2020-08-18:
Thanks for posting your error message!  Just trusting the error message, it looks like you have "main" missing from your luci config file.

Your luci config should start out with something like this:
```
config core 'main'
      option lang 'auto'
      option resourcebase '/luci-static/resources'
      option mediaurlbase '/luci-static/material'
```

I _think_ your bigger problem is that you can't putty in to fix it... so I don't know how would do any of that without just resetting the router.  I'm willing to help, but if you can't ssh and can't access the gui... I don't know of a simple way to access the router configs.

roshavagarga, 2020-08-18:
I might just be doing the putty thing wrong for all I know.

translucentfocus, 2020-08-18:
> 
> 
> I might just be doing the putty thing wrong for all I know.

[image]

If it looks like that, you should be able to access it.  If it doesn't even open a console... I'm afraid you'll just have to reset.

[Keep Settings](https://openwrt.org/docs/guide-quick-start/admingui_sysupgrade_keepsettings) should not be used when upgrading as you did... so I'm surprised your network is operating at all 🙂

Specifically:

"Only use it for the same firmware channel (release → release, snapshot → snapshot)."
"Cautiousness is mostly needed with major version upgrades, e.g. from 15.05 to 17.01, or from 15.05 to 18.06 etc."

feckert, 2020-08-19:
You have two option:
- Factory reset the router so you have a fresh installed system and reconfigure it again.
- Fix the luci config as @translucentfocus suggests and check if it works.

closed 2020-08-19 by feckert