openwrt/luci #4325: multiple switches with same port numbers

Parabol1337, 2020-08-01:
I've been using OpenWRT on my Mikrotik RB2011 for years now without an issue. But now I was upgrading amy device to 19.07. and I noticed strange behaviour with the two switches in this one.

It seems (l)uci isn't able to properly distinguish both configurations. When I add a vlan and configure a port to one of the switches, the port configuration on the other switch is also changed.So I have this issue now with multiple units.
The problem is only in LUCI and I guess it's because both switches have interfaces labeled 1~5 instead of what's written on the box. It looks like it's working normally, but when you submit changes it mixes up the configuration for both switches. Especially adding a VLAN to one switch will make it appear on the other and doesn't save the VID, instead just saves it as 3 sequentially. This can be fixed in the /etc/config/network file, but is garbled again when submitting on the Switch page in LUCI.

Parabol1337, 2020-08-03:
I have found the problem,

If i have 2 switches 
luci create 2 forms with the same ID, like 'cbi-network-switch_vlan' 

If i change this to cbi-network-switch0_vlan no vlans are loaded 😢😢😢😢😢😢

Parabol1337, 2020-10-18:
can everyone help?

erwann2000, 2020-12-16:
Hello,

I have the same issue with rb2011 since the 19.07 version.

I found another person who have this issue

https://forum.openwrt.org/t/mikrotik-rb2011-problem-with-multiple-switches/57185

I don't know if this issue  affects anothers devices with multiples switchs.

erwann2000, 2020-12-22:
When I only edit /etc/config/network and then I do "service network reload". It works like a charm.

jannoke, 2021-07-19:
I just hit my head with the same issue. Still present in 19.07.7

x19er, 2021-07-30:
I just hit this as well. I'm new to OpenWRT and was totally confused - the VLAN display in 19.07.07 LUCI isn't correct. Installed 18.06.9 and it all makes perfect sense and works as expected. In 18.06.9 I can see the three default VLANs on Switch0 and one on Switch1. In 19.07.07 there is only one VLAN for each Switch shown it won't let you tag/untag ports. If you do this in /etc/config/network any change is displayed on both Switches in LUCI.

ChristophFranzen, 2024-05-14:
I still see the same in 19.07.10 r11427-9ce6aa9d8d on an RB2011iLS with 2 built-in switches.

However, I had noc success yet creating a working configuration manually after Luci screwed it up.

How ist this supposed to work?

Can i have something like:

```
option device 'switch0'
option vlan '1'
option vid '1'
…
option device 'switch1'
option vlan '1'
option vid '1'
```

for the same VLAN on the 2 switches, or must the VLAN "numbers" be all non-equal (not only per switch), and the IDs both '1' if I want ports on both switches in the same VLAN?

jow-, 2024-05-15:
It should be no problem to configure the same vlan number on both switches, both the vid and the vlan number (table index) may be the same. In order to have them both act as one broadcast domain, you'd have to bridge both VLANs CPU ports in software though (by bridging `eth0.1` and `eth1.1` and configure layer 3 settings on the resulting `br-xxx` bridge)

ChristophFranzen, 2024-05-15:
Thank you for the answer, in the meantime I figured out the same by experimentation.
(Multiple VLAN numbers can have the same IDs, as it seems, but not vice versa. You MUST NOT configure VLAN 1 with IDs 1 AND 2 at another place, for example. => Parameters vid:vlan = 1:N.)

The startup problems of the network were not related (only) to VLAN switch settings, but also a typo at another place in the file I overlooked at first. I'm also not sure anymore that Luci really (always) writes the VID.

I tried to find a workaround to prevent Luci from damaging the configuration.

I've set up more VLANs than actually needed:
**eth0.1 = WAN**
**eth0.2 & eth1.3 bridged = LAN**
```
config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'
        option ar8xxx_mib_type '0'
        option ar8xxx_mib_poll_interval '500'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option vid '2'
        option ports '2 3 4 5 0t'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option vid '1'
        option ports '1 6 0t'

config switch_vlan
        option device 'switch0'
        option vlan '3'
        option ports '0t'

config switch
        option name 'switch1'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch1'
        option vlan '1'
        option vid '1'
        option ports '0t'

config switch_vlan
        option device 'switch1'
        option vlan '2'
        option vid '2'
        option ports '0t'

config switch_vlan
        option device 'switch1'
        option vlan '3'
        option vid '3'
        option ports '0t 1 2 3 4 5'
```
In this scenario, switch 1 only uses VLAN 3 for LAN, and switch 0 uses VLAN 1 for WAN and VLAN 2 for LAN.
Since VLANs 2 and 3 they use the same boadcast domain, and are "actually the same" (but numbered differently on the 2 switches).
The hope was that they can now being configured with Luci without interference, because on the first switch we would only change settings for VLANs 1 and 2, and on thw second only for VLAN 3.
It now turned out that Luci **reads** the VLAN "IDs" from the lines "option vlan '<number>'", and _NOT_ "option vid '<id-number>'". It also **prints** the VLAN lines for switch1 (the 2nd entry) as the configuration for **both** switches, so the 3 VLAN lines below each switch are identical, and reflect only the ones belonging to the 2nd switch1. The port images (which one is connected) seem to be mostly correct in this 3-VLAN-scenario, and reflect the actual swicht connections.

ChristophFranzen, 2024-05-15:
This is what Luci shows for the configuration above (of course only the "Luci label" is actually displayed):
**Format:**
_switch#: uci port #=Luci label=label on box case | …_
_vlan #: setting | …_

```
switch0: 0=CPU (eth0)= | 2=LAN 1=ETH2 | 3=LAN 2= ETH3 | 4=LAN 3=ETH4 | 5=LAN 4=ETH5 | 6=SFP=SFP | 1=WAN=ETH1
1: tagged | off | off | off | off | off | off
2: tagged | off | off | off | off | off | off
3: tagged | untagged | untagged | untagged | untagged | off | untagged

switch1: 0=CPU (eth1)= | 5=LAN 1=ETH10 | 4=LAN 2= ETH9 | 3=LAN 3=ETH8 | 2=LAN 4=ETH7 | 1=LAN 5=ETH6
1: tagged | off | off | off | off | off | off
2: tagged | off | off | off | off | off | off
3: tagged | untagged | untagged | untagged | untagged | off | untagged

```
Apart from the functional problem, that **has** to be corrected, there is also a "cosmetic" problem: having a web configuration with extremely misleading is not very useful.

I _think_ the following happens:

The "Luci labels" are probably generated from "/etc/board.json" and then named after the "roles" and ordered alphabetically.

This must become better. I suggest that Luci follows the order of appearance in board.json, and this should be generated correctly. Also a parameter like "label" is needed for the board.json file, where Luci can find the port designations printed on the box. This can be made optional, so Luci can revert to its old behaviour if no labels are defined (yet) for a known model. Luci should display a line with the "role" and above or below a line with the label, and not something labeled "$ROLE <automatically numbered _per switch_".

Here are the actual box labels from left to right at least for the RB2011iLS (and probably all other RB2011 models with SFP):

`| 2 LEDs | SFP | ETH1 ETH2 ETH3 ETH4 ETH5 | 10 LEDs | ETH6 ETH7 ETH8 ETH9 ETH10 |`

jow-, 2024-05-15:
Well, I don't think it will get fixed, afair it only affects multi-switch swconfig devices, devices with a single switch are unaffected and swconfig in general is deprecated and will get dropped in favor to dsa which will use descriptive netdev names.

ChristophFranzen, 2024-05-15:
For now, I've documented the switches as accurate as I could on the Wiki page: [https://openwrt.org/toh/mikrotik/rb2011#network_ports_switch_configuration], but I consider that to be a workaround until above mentioned problems are fixed.

ChristophFranzen, 2024-05-15:
For now, I've documented the switches as accurate as I could on the [Wiki page](https://openwrt.org/toh/mikrotik/rb2011#network_ports_switch_configuration) , but I consider that to be a workaround until above mentioned problems are fixed.

Someone has begun an unfinished ath79 port for the RB2011, so I personally wouldn't mind if this gets fixed there "automatically" for future releases, but having only a broken one isn't a good thing.