openwrt/luci #4331: Docker cannot be used

lloyd18, 2020-08-04:
[image]

[image]

@feckert

feckert, 2020-08-04:
You have to start first the docker service.
With `/etc/init.d/dockerd start`

lloyd18, 2020-08-04:
> You have to start first the docker service.
> With `/etc/init.d/dockerd start`
> 
> The Service is default enabled.

I tried. It didn't work.
[image]

feckert, 2020-08-04:
Could you please show me your `/tmp/dockerd/daemon.json` and your `/etc/config/dockerd` and your running docker process with `ps|grep docker` ?

lloyd18, 2020-08-04:
cat /tmp/dockerd/daemon.json

`{ "data-root": "\/opt\/docker\/", "log-level": "warn", "bip": "172.18.0.1\/24", "registry-mirrors": [ ], "hosts": [ ] }
`

cat /etc/config/dockerd

`
config globals 'globals'
	option data_root '/opt/docker/'
	option log_level 'warn'
	option hosts 'unix://var/run/docker.sock'
	option bip '172.18.0.1/24'
	option remote_endpoint '0'
`

ps|grep docker

` 9342 root      1096 S    grep docker
`

feckert, 2020-08-04:
Could you please change the /etc/config/dockerd config option hosts to:
`list hosts 'unix:///var/run/docker.sock'`
I think this is the problem.

I have also staged the following [P/R ](https://github.com/TDT-AG/packages/commits/pr/20200804-docker-ce) in my repository and this [commit](https://github.com/TDT-AG/packages/commit/70ecc1a4f08a3a1f4d046e03c7d026edb22122c6) should fix your issue.

lloyd18, 2020-08-04:
Thank you for your help. I merged your pull requests, and recompiled the firmware, but the problem still exists.

closed 2020-08-13 by lloyd18