openwrt/luci #4348: Disable openwrt luci auto refresh

mh4nx7net, 2020-08-12:
maybe it's trivial issues
but, idk where i can get the config for disable/ change refresh interval luci auto refresh persistently
whenever i have refresh luci webUI. it's autorefresh just come up again.

thanks for the attention's
`here what may look like`
[image]

`just trying to figure out what an any word that can be possible; LOL 🤣 `

[image]

jow-, 2020-08-16:
This all happens in client side code. You would need to edit `luci.js` and modify the behaviour of the [`LuCI.poll` class](http://openwrt.github.io/luci/jsapi/LuCI.poll.html)

closed 2020-08-16 by jow-

mh4nx7net, 2020-08-19:
thanks, help alot