openwrt/luci #4352: mwan3: Interfaces grid can become mis-aligned on MWAN3 interface status page

jamesmacwhite, 2020-08-13:
Depending on the content within each interface block, it can cause the alignment to be broken. I believe this could be due to the variable height on each grid block when the interface uptime changes every few seconds.

[image]

Potentially a CSS tweak is needed, possibly to force the height on each interface block to avoid the issue.

feckert, 2020-08-14:
I don't like the CSS thing. But it was like that when I took over the code. I tend to be more of a table virw. With the CSS we will always have problems, even if we fix it now. It may look different and destroy the view on another theme!
@jamesmacwhite If you like you can then convert this into a table?

aaronjg, 2020-08-18:
What if we change the space after the colon to be a non-breaking space, or set a css-property to disable line wrapping?

PR #4361 (2020-08-18): luci-app-mwan3: fix interface grid
Interface grid can become misaligned due to varying number of
lines. Use non-breaking spaces to ensure all interface boxes are three
lines long.

Fixes #4352
closed 2020-08-18 by feckert