openwrt/luci #4354: "No password set!"-message suggests that SSH is disabled, while it is enabled

skleeschulte, 2020-08-15:
When accessing OpenWrt's web-interface after a fresh install, the following message is shown: "No password set! There is no password set on this router. Please configure a root password to protect the web interface and enable SSH."

[image]

This suggests that SSH is not enabled until a root password is set. Nonetheless, I can successfully connect to OpenWrt using Putty/SSH/user root and an empty password.

LuCI openwrt-19.07 branch (git-20.136.49537-fb2f363) / OpenWrt 19.07.3 r11063-85e04e9f46

castillofrancodamian, 2020-08-15:
You're right, it should be removed that SSH is not enabled.

hnyman, 2020-08-16:
That is probably a leftover from time when initially only telnet was enabled, and after the password was set, then telnet got disabled and ssh got enabled. 

That practice was dropped in 2015 by https://github.com/openwrt/openwrt/commit/d196b1fc2eddf7cbab4351452b6fc9b41f15b142

stangri, 2020-08-16:
Looks like it's a text in the header file for all the themes. If I were to fix it, would I also need to submit a new templates pot file or would it be rebuilt automatically?

hnyman, 2020-08-16:
.pot gets refreshed via i18n-sync. It should be submitted at the same time, as it makes no sense to leave that unpatched.

I will patch themes and will try to fix as many translations as I can...

hnyman, 2020-08-16:
Fixed in master by 258e74c1c3e46 and 641c3b88aef

closed 2020-08-16 by hnyman

hnyman, 2020-08-16:
Backported to 19.07 with f3dc664144 and 97553f8065