openwrt/luci #4356: How to view Active clients on the LAN in OpenWrt?

guilhermetb, 2020-08-16:
Hello, I would like help if possible.

I used DD-WRT for a long time and recently switched to OpenWRT.
How can I activate this feature that I am using as an example of the DD-WRT printscreen (Staus -> LAN -> Active Clients) in OpenWRT?

[image]

guilhermetb, 2020-08-16:
192.168.1.1 - TL-WR1043ND v1 - OpenWrt 19.07.3 - XX:XX:XX:XX:7E:86
192.168.1.2 - TL-WA850RE - Repeater tplink - XX-XX-XX-XX-1D-79
[image]

guilhermetb, 2020-08-16:
I would like information from only the hosts that are exchanging packets on the network at the moment and not the leases granted by dhcp.

Is there any way to install a package through opkg that gets the information below in real time as in dd-wrt?

Host | Ip | Mac | Interface | Cont. | Proportion [4096]

I will give an example of the feature that I miss in relation to DD-WRT, in it through the path "Status -> LAN"

What comes closest to this feature in OpenWRT is through the path "Status -> Realtime Graphs -> Traffic -> br-lan"

However in OpenWRT it does not show the ips and hosts that are using the network in real time.

Can I get this feature in OpenWRT?

[image]

jow-, 2020-08-16:
This is not possible.

closed 2020-08-16 by jow-

guilhermetb, 2020-08-16:
> This is not possible.

why?