openwrt/luci #4363: No Graphs showing in new install

mikerog25, 2020-08-18:
Model	Netgear WNR3500L
Architecture	Broadcom BCM4716
Firmware Version	OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.229.71287-57456a7
Kernel Version	4.14.180

Chosing "Realtime Graphs",  page shows,  but no graphics on any of the tabs..

In SYSLOG and Kernel Log  this shows , while page is open 

Tue Aug 18 19:35:42 2020 kern.warn kernel: [40863.914486] __do_page_fault: 2 callbacks suppressed
Tue Aug 18 19:35:42 2020 kern.info kernel: [40863.914504] do_page_fault(): sending SIGSEGV to luci-bwc for invalid write access to 00000000
Tue Aug 18 19:35:42 2020 kern.info kernel: [40863.929119] epc = 77ef911c in libc.so[77edc000+98000]
Tue Aug 18 19:35:42 2020 kern.info kernel: [40863.934448] ra  = 77e9e1c5 in libiwinfo.so[77e9a000+1c000]

Havent found any other examples of this , so raising issue 

Updated all software , just in case fixed.    Still happeneing 

No workarounds found..

jow-, 2020-08-20:
Using broadcom-wl ?

mikerog25, 2020-08-26:
Hi, yes  it is Broadcom but I have both radios disabled, using other WiFi solutions on network.  I rebooted a couple of times without luck, but then detached it from wan connection to try a downgrade.  When booting up offline to do the downgrade to 18.07, graphs were working.   How weird.   When I get a chance it try it again connected .    I guess I'm happy for issue to close as its not blocking me as I'm using another device for now.

Nibi-git, 2021-03-15:
Model D-Link DSL-2650U/BRU/D 
Architecture Broadcom bcm63xx
Firmware Version OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-21.070.25463-b4025cf
Kernel Version 4.14.221

Same bug vith empty Graphs. If I disable Wan interface  (3G USB modem) - the graphs start working. 

```
Mon Mar 15 20:04:23 2021 kern.info kernel: [149940.235072] do_page_fault(): sending SIGSEGV to luci-bwc for invalid write access to 00000000
Mon Mar 15 20:04:23 2021 kern.info kernel: [149940.244063] epc = 77e875cc in libc.so[77e6a000+98000]
Mon Mar 15 20:04:23 2021 kern.info kernel: [149940.249509] ra  = 77e27328 in libiwinfo.so[77e22000+20000]
Mon Mar 15 20:04:24 2021 kern.info kernel: [149940.678462] do_page_fault(): sending SIGSEGV to luci-bwc for invalid write access to 00000000
Mon Mar 15 20:04:24 2021 kern.info kernel: [149940.688179] epc = 77f095cc in libc.so[77eec000+98000]
Mon Mar 15 20:04:24 2021 kern.info kernel: [149940.694495] ra  = 77ea9328 in libiwinfo.so[77ea4000+20000]
```