openwrt/luci #4370: Insufficient permissions to read UCI configuration.

eliaspizarro, 2020-08-19:
Steps:
1
[image]

2
[image]

noyidoit, 2021-06-26:
Is there any tutor or docs explained how to write `rpcd/acl.d/xxx.json` ?
My only workaround is revert the `map.html` to a old version, that's so uncivilized.

n3gwg, 2021-12-08:
Has there been any movement on squashing this bug at all?
What ostensible workarounds have been tested thus far?

I read in a forum a patch was submitted but it was rejected and no explanation given as to why.

stangri, 2021-12-09:
@n3gwg you may want to test @dibdot's fix: https://github.com/dibdot/luci/commit/d915425c01651f13ab7ba74a72ccbd229433a87b

commit 18b11307 (2022-01-25): luci-app-openvpn: fix stray uci permission warning

The OpenVPN file view uses a dummy Map() instance to render the breadcrumb
template which triggers a uci permission error since the view is being
rendered by a form() action which does not set up the expected permission
flags. CBI Map() instances should only be used for cbi() dispatch targets.

Solve the issue by appending the breadcrumb template directly to the
SimpleForm() instance and by removing the redundant dummy Map() instance.

Fixes: #4370
commit e927a11d (2022-01-25): luci-app-openvpn: fix stray uci permission warning

The OpenVPN file view uses a dummy Map() instance to render the breadcrumb
template which triggers a uci permission error since the view is being
rendered by a form() action which does not set up the expected permission
flags. CBI Map() instances should only be used for cbi() dispatch targets.

Solve the issue by appending the breadcrumb template directly to the
SimpleForm() instance and by removing the redundant dummy Map() instance.

Fixes: #4370

(cherry picked from commit 18b1130711b6ff7aacba98034e2bd7f4893b33df)
commit f3debdcf (2022-01-25): luci-app-openvpn: fix stray uci permission warning

The OpenVPN file view uses a dummy Map() instance to render the breadcrumb
template which triggers a uci permission error since the view is being
rendered by a form() action which does not set up the expected permission
flags. CBI Map() instances should only be used for cbi() dispatch targets.

Solve the issue by appending the breadcrumb template directly to the
SimpleForm() instance and by removing the redundant dummy Map() instance.

Fixes: #4370

(cherry picked from commit 18b1130711b6ff7aacba98034e2bd7f4893b33df)
closed 2022-01-25 by jow-