openwrt/luci #4372: how to use uci api command with javascript

mh4nx7net, 2020-08-19:
[image]
iam so sorry if this question is strictly dumb enough, just idk how to start

[image]
that's it,  when i try to build application on top of luci with route controller redirectet to this files, whereas `coba_reg.js` as my main  `page application of..`
`/www/luci-static/resources/view/autowms/coba_reg.js`
i think i need to talk with LuCI.uci.* application interface for call of each value inside of `/etc/config/myconf`
but i dont really know how to call them..
can you, or anybody to help me
thanks alot sirr

nb. I just cheated a little of `/www/luci-static/resources/view/network/interfaces.js` but don't really get it
then, what iam ask really is.
how can i develop/test each of script that's work or not with browser console ? there's any way to do it ?

jow-, 2020-08-19:
Your view needs a `'require uci';` statement at the top to request the class. Before you can work with uci data, you need to load it. Usually it makes sense to do that in the load procedure of a view to avoid the need to nest all your logic into a Promise.then block:

```
'use strict';
'require uci';
'require view';

return view.extend({
    load: function() {
        return Promise.all([
            uci.load('autowms'),
            uci.load('something_else')
        ]);
    },

    render: function() {
        var keypatch = uci.get('autowms', 'status', 'keypatch');

        return E('code', [
            'The value of autowms.status.keypatch is %s'.format(keypatch)
        ]);
    }
});
```

To debug uci operations in the browser console you cannot use the `require` magic, you need to use the following code:
```
L.require('uci').then(function(uci) { return uci.load('autowms').then(function() { console.debug(uci.sections('autowms')); }); });
```

closed 2020-08-19 by jow-

mh4nx7net, 2020-08-19:
thanks, clear enough 

sorry if iam asking some about trivial thing.
thankyou so much 
[image]