openwrt/luci #4389: luci-app-mwan3: Check quality of link failure_loss and recovery_loss values do not match defaults in mwan3track

jamesmacwhite, 2020-08-25:
@feckert 

I was updating the documentation for mwan3 https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3 as I noticed the interface configuration was missing the check_quality configuration options, I then found the default values used for failure_loss and recovery_loss are mismatched in mwan3track and LuCI.

Defaults used in mwan3track:

https://github.com/openwrt/packages/blob/master/net/mwan3/files/usr/sbin/mwan3track#L151-L152

Defaults in LuCI:

`failure_loss` defaults to 40 in mwan3track, default is 20 in LuCI:

* https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interfaceconfig.lua#L122

`recover_loss` defaults to 10 in mwan3track, default is 5 in LuCi:

* https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interfaceconfig.lua#L143

Should LuCI be updated to be consistent? I guess it's not a major problem.

Also is `check_quality` meant to be written the interface config when it is enabled in LuCI? If so this doesn't seem to happen but the four check quality options are.

aaronjg, 2020-08-28:
Yes. Luci should be updated to use the mwan3track defaults, and if the default is unset in /etc/mwan3/config, luci should not set it unless the user chooses a value.

feckert, 2020-08-31:
@jamesmacwhite thanks for the hint.
I have fixed this in the last commit of luci-app-mwan3.

closed 2020-08-31 by feckert

jamesmacwhite, 2020-08-31:
@feckert Thanks. Documentation has been updated to match, as it was missing entirely from the interface configuration table.

https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#interface_configuration

aaronjg, 2020-08-31:
@feckert, I think even with this fix, luci will add the default value to mwan3 config if it has not been set. Is that the desired behavior? This seems non standard to me. For example the dnsmasq luci app leaves unset config options unset.

feckert, 2020-08-31:
As long as we do not enable the option [check_quality ](https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interfaceconfig.lua#L104) then the depending values are not saved into the running configuration.

commit 8e4edd20 (2020-08-31): luci-app-mwan3: sync default values with mwan3

fixes #4389