openwrt/luci #4390: With wpad-basic-wolfssl ieee 802.11w option is missing

howl, 2020-08-27:
OpenWrt Master changed the last week from wpad-[basic|mini] to wpad-basic-wolfssl. The reason was to support SAE/WPA3-Personal out of the box.

In LuCI this works fine but lacks the option to configure ieee 802.11w even when WPA2 or WPA2/WPA3 is used. I'm not sure if with WPA3 only should appear as if I'm not wrong WPA3 enforces it. Then at least when WPA3 only mode is not selected it should appear.

Despite the option is not shown I have tested both WPA2 and WPA2/WPA3 modes configuring "option ieee80211w" to its three possible values and it plays fine with wpad-basic-wolfssl. Even when using mixed mode it has it's effect for the WPA2 part.

jow-, 2020-08-27:
Not a LuCI bug. See https://github.com/openwrt/openwrt/pull/3347

closed 2020-08-27 by jow-

howl, 2020-08-27:
Sorry.