openwrt/luci #4395: luci-ssl: change the default SSL lib to wolfssl?

hnyman, 2020-08-28:
@jow- 

It seems to me that the main OpenWrt has now adopted wolfssl as the default SSL library since commit
https://github.com/openwrt/openwrt/commit/a14f5bb4bd263c21e103f13279d0c2ff03e48fe5

Should we now change also LuCI to use wolfssl be default instead of mbedtls ?

https://github.com/openwrt/luci/blob/5c1f36ceb5c8ccd76bd61e170dd42748d502ba0d/collections/luci-ssl/Makefile#L13

hnyman, 2020-08-29:
Reference to OpenWrt PR
https://github.com/openwrt/openwrt/pull/3363
"Add px5g WolfSSL version"

aparcar, 2020-08-30:
Related https://github.com/openwrt/luci/pull/4401

hnyman, 2020-08-30:
Closing this, as aparcar did the PR #4401, where the issue can be handled.

closed 2020-08-30 by hnyman