openwrt/luci #4405: Luci package offline

knobik, 2020-09-01:
https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a72/luci/Packages

Its empty.

hnyman, 2020-09-01:
Global build problem (when using SDK like buildbot).

All packages have failed to build in the buildbot...
(faillogs:  https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a15_neon-vfpv4/ )

Like this:
```
fakeroot: preload library `libfakeroot.so' not found, aborting.
../../luci.mk:298: recipe for target '/builder/shared-workdir/build/sdk/bin/packages/arm_cortex-a15_neon-vfpv4/luci/luci_git-20.113.70804-9d02ef1_all.ipk' failed
make[3]: *** [/builder/shared-workdir/build/sdk/bin/packages/arm_cortex-a15_neon-vfpv4/luci/luci_git-20.113.70804-9d02ef1_all.ipk] Error 1
time: package/feeds/luci/luci/compile#0.39#0.14#0.74
```

Possibly fixed with 
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=5587f19c362a805636da91d9d550928b4a2998fc
and
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=729a75c3b2db0aa25745e8db425711cc761fa1f6

But it will take some time to trickle down to packages building.

Try again in 1-2 days.

jow-, 2020-09-08:
Not a LuCI specific bug

closed 2020-09-08 by jow-