openwrt/luci #4412: Question: ACL for form.FileUpload widget?

dibdot, 2020-09-04:
Hi,
I've nearly finished the mgration of travelmate to client side JS and currently I'm looking in remaining small issues. 
One of them is ACL/FileUpload related. I'm unable to control/filter which files gets listed in this widget (I'm only interested in files with the extension "*.login"), e.g.

Source:
```
[...]
		o = s.taboption('travelmate', form.FileUpload, '_script', _('Auto Login Script'),
			_('External script reference which will be called for automated captive portal logins.'));
		o.root_directory = '/etc/travelmate';
		o.enable_remove = false;
		o.enable_upload = false;
		o.modalonly = true;
		o.uciconfig = 'travelmate';
		o.ucisection = 'uplink';
		o.ucioption = 'script';
[...]
```

ACL:
 ```
[...]
		"read": {
			"cgi-io": [ "exec" ],
			"file": {
				"/etc/travelmate/*.login": [ "list" ],
[...]
```
But the widget still shows all files within the given directory  ... furthermore it's also working without any ACL - I can browse this directory and select files!?
I'm a bit lost here ...

Many thanks for any valuable hint!

jow-, 2020-09-07:
- Listing / browse access to all locations is allowed by default yet (https://github.com/openwrt/luci/blob/master/modules/luci-base/root/usr/share/rpcd/acl.d/luci-base.json#L14)
- There is no option support for filtering the file list yet, but the (untested) code below should achieve what you need:

```javascript
o = s.taboption('travelmate', form.FileUpload, '_script', _('Auto Login Script'),
			_('External script reference which will be called for automated captive portal logins.'));
...
/* Overwrite the renderWidget() implementation to obtain the internal filebrowser widget ... */
o.renderWidget = function(section_id, option_index, cfgvalue) {
	var browserEl = new ui.FileUpload((cfgvalue != null) ? cfgvalue : this.default, {
		id: this.cbid(section_id),
		name: this.cbid(section_id),
		show_hidden: this.show_hidden,
		enable_upload: this.enable_upload,
		enable_remove: this.enable_remove,
		root_directory: this.root_directory,
		disabled: (this.readonly != null) ? this.readonly : this.map.readonly
	});

	/* Monkey patch the internal filebrowser widget to filter the list of files to display */
	browserEl.renderListing = function(container, path, list) {
		return ui.FileUpload.prototype.renderListing.apply(this, [
			container, path,
			/* Only keep directory entries or file entries ending with ".login"
			   Note that this will skip symlinks, fifos etc. too */
			list.filter(function(entry) {
				return ((entry.type == 'directory') || 
				        (entry.type == 'file' && entry.name.match(/\.login$/)));
			})
		]);
	};

	return browserEl.render();
};
```

dibdot, 2020-09-07:
Thanks for the explanation and the proposed workaround - this works outstandingly well! :+1: :1st_place_medal: 

Great stuff!

closed 2020-09-07 by dibdot