openwrt/luci #4416: QoS over Nftables : Ipaddr

DrKinSlayeR, 2020-09-07:
Hello everyone,

Sorry for my english, I do my best to be understandable.
I would like changing the IP address column by hostname. That's possible or not? I've many peripherals connected, and that will be more simple for me with hostname.

Thanks for your help!

DrKinSlayeR, 2020-09-14:
HELP ? :D

feckert, 2020-09-14:
As I understand your question, that is not possible. Please can you provide more information. Which LuCI module is it? Which OpenWRT version do you use.

DrKinSlayeR, 2020-09-14:
[image]
[image]

I use luci-app-nft-qos and nft-qos. I would like changing ip address to hostname or add a column with the hostnames. If is that possible.

DrKinSlayeR, 2020-09-23:
help @feckert :'(

adelton, 2025-01-11:
Since it took me a while to figure out where this table is in the OpenWrt 24.10.0-rc5 (r28304-6dacba30a7) UI, for whoever might decide to take this: after installing `luci-app-nft-qos` in `System > Software`, I had to `Log out` and log back in. After that in `Status > Realtime Graphs` there is a new tab `Rate`. For something to show up in those tables, things would likely need to be configured in `Services > QoS over Nftables` -- for me the two tables on that page always have `Collecting data...`: 
[image]

Looking at the tab `Connections`, there is a button `Enable DNS lookups` there, so something similar likely could be done for the `Rate` tables.