openwrt/luci #4419: Archer C7 v2 - wlan0 and wlan1 fail to enable after sometime of running

isrizil, 2020-09-08:
Hello,
after I enabled the Wlan0 and Wlan1 interfaces it run for several hours and than get into cycle of -disable, enable, disable, enable

attached log from the system log

Thanks
using - OpenWrt 19.07.3 r11063-85e04e9f46
[openwrt_archer_c7_v2.txt](https://github.com/openwrt/luci/files/5186972/openwrt_archer_c7_v2.txt)

jow-, 2020-09-08:
Not a LuCI specific bug.

closed 2020-09-08 by jow-

isrizil, 2020-09-08:
hey, what does it mean? i am new to openwrt?

feckert, 2020-09-08:
Only the Webgui is covered here.
If you have a problem with the base system then you have to open a ticket [flyspray](https://bugs.openwrt.org/) or in the [package](https://github.com/openwrt/packages/issues) feed.