openwrt/luci #4425: Process list misses processes with large VSZ

dsheets, 2020-09-10:
Busybox `top` outputs an 'm' suffix (for MiB) if the byte size necessary is greater than 100000 KiB. See <https://github.com/brgl/busybox/blob/master/procps/top.c#L698>.

The luci regular expression that matches top output lines is at <https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/sys.lua#L382>. Notably, this regex doesn't match a line with a VSZ field having an 'm' suffix.

hnyman, 2020-09-10:
Sis you test which string would work?

`"^ *(%d+) +(%d+) +(%S.-%S) +([RSDZTW][<NW ][<N ]) +(%d+[m]) +(%d+%%) +(%d+%%) +(.+)"`  ????

dsheets, 2020-09-14:
I have not yet investigated patching luci or any component of OpenWRT and I have very limited time to pursue OpenWRT/luci hacking. I am not familiar with the regex dialect in use here but I wonder if `[m]` will require one of the singleton set `{'m'}` to be present (i.e. will not match the non-'m' case)? Maybe `m?` would be appropriate?

hnyman, 2020-09-17:
@jow- 
Would 
`^ *(%d+) +(%d+) +(%S.-%S) +([RSDZTW][<NW ][<N ]) +(%d+m*) +(%d+%%) +(%d+%%) +(.+)` ????
be ok ?

(not sure how that possible "m" should be written.  `m*`   or `[m]*`  or ???)

jow-, 2020-09-17:
Something like `...(%dm?)...` should do. Question is whether there are other possible units. I suppose theoretically there could be `g` and `e` as well.

commit 4f2ecd88 (2020-09-19): luci-base: tweak busybox top regex to accept 'm' in large VSZ

Tweak the lua regex string for process list, so that processes
with extremely large memory sizes (over 100 MB) are matched.
For those large processes busybox top shows a MB value like '234m'
instead of a normal kB value like 234000.

Fixes #4425

Reference to https://github.com/openwrt/luci/issues/4425#issuecomment-694073479
commit 8aceafe4 (2020-09-19): luci-base: tweak busybox top regex to accept 'm' in large VSZ

Tweak the lua regex string for process list, so that processes
with extremely large memory sizes (over 100 MB) are matched.
For those large processes busybox top shows a MB value like '234m'
instead of a normal kB value like 234000.

Fixes #4425

Reference to https://github.com/openwrt/luci/issues/4425#issuecomment-694073479

(cherry picked from commit 4f2ecd88fd02c1279041f45ccd758184b153c875)
commit 8401f59e (2020-09-19): luci-base: tweak busybox top regex to accept 'm' in large VSZ

Tweak the lua regex string for process list, so that processes
with extremely large memory sizes (over 100 MB) are matched.
For those large processes busybox top shows a MB value like '234m'
instead of a normal kB value like 234000.

Fixes #4425

Reference to https://github.com/openwrt/luci/issues/4425#issuecomment-694073479

(cherry picked from commit 4f2ecd88fd02c1279041f45ccd758184b153c875)
closed 2020-09-19 by hnyman