openwrt/luci #4427: Openwrt autorefresh time change?

kissmorning, 2020-09-10:
hello
I want to change the update interval to 1 second ( Openwrt Changes Pool status )
how can I do it?
[image]
[image]

feckert, 2020-09-14:
I think it was possible to configure this in the uci before we moved to Javascript but this was discarded or not ported.
In the source we can set this in the [source]( https://github.com/openwrt/luci/blob/09014e01ab570a7b73d6b3f5d2f1a88bf7a763b4/modules/luci-base/htdocs/luci-static/resources/luci.js#L973) for each poll.

jow-, 2020-09-14:
As written above, this needs to be [changed in the code](https://github.com/openwrt/luci/blob/master/modules/luci-mod-status/htdocs/luci-static/resources/view/status/load.js#L15).

closed 2020-09-14 by jow-

kissmorning, 2020-09-14:
In which folder should I make this change? I'm using winscp

kissmorning, 2020-09-14:
http://192.168.1.1/cgi-bin/luci/admin/status/overview

i need this website overview 1second setting