openwrt/luci #4428: luci-app-mwan3: Misleading "Interface has a higher reliability requirement than tracking hosts" warning

jamesmacwhite, 2020-09-11:
@feckert 

If the `reliability` option is not directly specified on a mwan3 interface config, LuCI shows this warning:

_WARNING: Interface wanX has a higher reliability requirement than tracking hosts (4)_

Example that triggers this warning:

```
config interface 'wanX'
	option enabled '0'
	list track_ip '8.8.4.4'
	list track_ip '8.8.8.8'
	list track_ip '208.67.222.222'
	list track_ip '208.67.220.220'
	option family 'ipv4'
```

However if I specify `option reliability '1'` on this interface the warning goes away, however `1` is the default, so it shouldn't be an issue in the first place. It is even displayed in the table, when not directly specified. I think the warning is misleading, as the reliability value isn't higher than the tracking hosts count at all.

jamesmacwhite, 2020-09-22:
@feckert I don't know what your thoughts are on this one. Defining reliability across any interface, even if it's the same as the default isn't exactly a bad thing, but it was discovered when looking to reduce/simplify the default configuration.

feckert, 2020-09-22:
I think we should fix the code on this [line](https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interface.lua#L102). Because we removed the option from uci the check fails and so the `overview[iface]["reliability"] ` is always false. 

We should change the line from

`local reliabilityNumber = tonumber(uci:get("mwan3", iface, "reliability"))`

to 

`local reliabilityNumber = tonumber(uci:get("mwan3", iface, "reliability"))  or "1"`

jamesmacwhite, 2020-09-22:
That would work as it provide a numeric value as a fallback so should avoid false errors?

feckert, 2020-09-22:
Fixed with commit https://github.com/openwrt/luci/commit/ab5b05dc2587323b6df971d9d6cf23844132571a

closed 2020-09-22 by feckert