openwrt/luci #4430: Support not sending any Host Name in DHCP packets as DHCP client

macman31, 2020-09-11:
I would like my OpenWRT router to **not** send any Host Name ([DHCP option 12](https://tools.ietf.org/html/rfc2132#section-3.14) in DHCP packets) when acting as a DHCP client. This is supported by OpenWRT but it is not supported in LuCI.

The current behavior is that OpenWRT defaults to [sending the system hostname](https://github.com/openwrt/openwrt/blob/14024f6e0fe883ae413b87391121bb6e221c9312/package/network/config/netifd/files/lib/netifd/proto/dhcp.sh#L48) if none has been defined in LuCI in an interface configuration, tab "General Settings", option `Hostname to send when requesting DHCP` (for the `DHCP client` protocol).

If a hostname is entered in `Hostname to send when requesting DHCP`, then LuCI checks that it is indeed a valid hostname before saving the configuration:

https://github.com/openwrt/luci/blob/a314b593ec1de84ce0293e18010405ff1ae5ccf6/modules/luci-base/htdocs/luci-static/resources/protocol/dhcp.js#L22-L29

https://github.com/openwrt/luci/blob/a314b593ec1de84ce0293e18010405ff1ae5ccf6/modules/luci-base/htdocs/luci-static/resources/validation.js#L348-L358

This causes an issue when wanting to **not** send any DHCP Host Name: if the field is left empty then the system hostname will be sent, and if it is not empty then the value entered will be sent as the hostname.

OpenWRT solves this problem by allowing the use of `option hostname '*'` for an interface configuration to mean ["do not send any hostname"](https://github.com/openwrt/openwrt/blob/14024f6e0fe883ae413b87391121bb6e221c9312/package/network/config/netifd/files/lib/netifd/proto/dhcp.sh#L49).

However when trying to set `Hostname to send when requesting DHCP` to `*` in LuCI, the web interface tries to validate the option as an hostname and it refuses to save the interface configuration because `*` is an invalid hostname.

Here are two ideas on how to solve this issue:
1. Find a way to allow `*` to be entered in `Hostname to send when requesting DHCP`, and add a tooltip/description indicating that `*` means "Do not send any hostname". The challenge is that it cannot be done easily by modifying the `hostname` datatype in `validation.js` (see snippet above), as this datatype could also be used elsewhere.
2. Add a checkbox near `Hostname to send when requesting DHCP` that would be used to indicate that no hostname should be sent (and it would grey out `Hostname to send when requesting DHCP`). If checked, it would save `*` as the value of `option hostname`, so that OpenWRT knows that it shouldn't send any hostname.

closed 2020-09-16 by jow-

feckert, 2020-09-16:
@jow I think we should also update the hint?
Otherwise nobody knows that this is also possible. I wouldn't have known that if I wasn't looking at the source.

jow-, 2020-09-16:
Feel free to

commit bdcd0949 (2020-09-16): luci-base: dhcp.js: allow hostname to be "*"

Fixes: #4430
commit 14693000 (2020-09-16): luci-base: dhcp.js: allow hostname to be "*"

Fixes: #4430

(cherry picked from commit bdcd09496342d3b8609e718e289656427fa0b28c)