openwrt/luci #4431: https-dns-proxy not show in luci

hendysama, 2020-09-11:
help me how to fix... thx

/usr/lib/lua/luci/model/cbi/https-dns-proxy.lua:57: attempt to index local 'value' (a nil value)
stack traceback:
	/usr/lib/lua/luci/model/cbi/https-dns-proxy.lua:57: in function 'getProviderName'
	/usr/lib/lua/luci/model/cbi/https-dns-proxy.lua:98: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1340: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1023: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>
[image]

stangri, 2020-09-12:
I believe I need to fix ACL asap. I'll try to submit a patch this weekend.

stangri, 2020-09-13:
It's not an ACL issue, that was fixed a while ago. Can you walk me thru exact steps how did you arrive to this issue? What is the OpenWrt version? How and where from did you install packages? Is the https-dns-proxy service installed/running?

stangri, 2020-09-20:
@hendysama did you resolve this issue?

stangri, 2020-09-24:
Closing due to lack of response.

closed 2020-09-24 by stangri