openwrt/luci #4436: dashboard wifi pane is crowded when using travelmate

nwf, 2020-09-16:
`travelmate` creates a lot of uci stanzas in `wireless` (one per network it knows how to roam between) but it keeps most of them `disabled` most of the time.  It'd be nice if the dashboard view did not show (or could be configured not to show) these `disabled` networks.

I see, skimming the source, that there are checks for `.visible` of both radios and devices, but I don't know what would make that attribute falsey.

jow-, 2020-09-16:
Can you please attach a screenshot illustrating the problem?

nwf, 2020-09-16:
Here's an example with two AP radios and one client radio with three possible settings.  The two client configurations that are not active are also disabled in `uci` and it would be nice if they were not shown.

[image]