openwrt/luci #4451: luci-mod-network: No values displayed with odhcp - OpenWrt 19.07.4

chemicalstorm, 2020-09-20:
Hi,

On `OpenWrt 19.07.4 r11208-ce6496d796` with `luci-mod-network git-20.261.50352-d98fff3-1`, the "DHCP and DNS" page is only displaying the "Server Settings" section with the following message:
This section contains no values yet

See 
[image]

Looking at the network console, I can see leases being returned by the periodic calls to the `/cgi-bin/luci/admin/ubus` endpoint with the following body: `[{"jsonrpc":"2.0","id":64,"method":"call","params":["<TOKEN>","luci-rpc","getDHCPLeases",{}]}]`.
There is no error displayed in the console and I tried to step-by-step the execution and processing and validation of the lease looked fine.

My understanding is that the page does not have the `lease_status_table` in its DOM (created in the `CBILeaseStatus` variable) and thus processing of lease is done but nothing gets updated.

Here is the DOM for the `#view` `div`:
```
<div id="view"><div id="cbi-dhcp" class="cbi-map flash" data-idref="18fce7d7"><h2 name="content">DHCP and DNS</h2><div class="cbi-map-descr">Dnsmasq is a combined <abbr title="Dynamic Host Configuration Protocol">DHCP</abbr>-Server and <abbr title="Domain Name System">DNS</abbr>-Forwarder for <abbr title="Network Address Translation">NAT</abbr> firewalls</div><div id="cbi-dhcp-dnsmasq" class="cbi-section" data-idref="cbcbbfc0"><legend>Server Settings</legend><em>This section contains no values yet</em><br><br></div></div><div class="cbi-page-actions control-group"><div class="cbi-dropdown btn cbi-button cbi-button-apply important" tabindex="0" data-idref="1bbaa588" data-changed="true"><ul tabindex="-1" style="" class=""><li data-value="0" selected="" display="0" class="">Save &amp; Apply</li><li data-value="1">Apply unchecked</li></ul><span class="more" tabindex="-1">···</span><span class="open" tabindex="-1">▾</span><div><input type="hidden" value="0"></div></div> <button class="cbi-button cbi-button-save">Save</button> <button class="cbi-button cbi-button-reset">Reset</button></div></div>
```

I tried to debug the creation of the DOM but I did not manage to find the proper functions that handle it.

Let me know what additional info you'd need to assist on that issue!

Thanks.

jow-, 2020-09-20:
This happens if no `config dnsmasq` section is present.

closed 2020-09-20 by jow-

chemicalstorm, 2020-09-20:
Thanks! So that means that support for an `odhcpd`-only install has been dropped?
If so, would there be interest for a patch if I manage to bring back the lease section?

jow-, 2020-09-20:
I think it never was officially supported and only worked by accident, a patch would be much appreciated, yes :)