openwrt/luci #4452: Creating new bridged interface gives error "network device is not present"

tjeulink, 2020-09-20:
When i create a new interface through luci, bridging two wifi networks, it results in the interface throwing an error because luci does not create the interface correctly apparently.

steps to reproduce: create two wlan networks (for me this is on two different chipsets, 2.4ghz and 5.8ghz). then go to the network>interfaces and click create new interface. set protocol to static, and bridge the two wlan networks just created. click create. observe the error from the luci interfaces page.

jow-, 2020-09-20:
Not a LuCI issue. the error you see is emitted by netifd.

closed 2020-09-20 by jow-

tjeulink, 2020-09-20:
Where should i file the bugreport for netifd? sorry i'm not too good at these types of things. people on reddit told me it was a wrong configuration from my UCI export so i assumed it was luci that misconfigured it.

jow-, 2020-09-20:
I am not convinced that there actually is a bug. I just tested it locally and things work as expected. The error you quoted can happen for any number of reasons, e.g. when attaching a wifi network to multiple bridges, or when configuring an unavailable channel.

You could ask at https://forum.openwrt.org/ for further help.