openwrt/luci #4467: Menu after logging in

thedyerman, 2020-09-23:
Menu is causing some type of timeout when first logging into LUCI interface. 
Running commit c17ec09 but has been an issue for the last few.

[image]

Easily recreate it by logging out and back into interface (default or material theme), happens every time across multiple routers I have running (all MediaTek although I'm not sure that's relevant)

rmilecki, 2020-09-28:
According to my bisecting & testing this issue is caused by the [5e9c23c6f40f ("client: allow keep-alive for POST requests")](https://git.openwrt.org/?p=project/uhttpd.git;a=commitdiff;h=5e9c23c6f40ff26209ef22cfeeda4904a5918f3d).

I don't know much about keep-alive, it seems like some problem with `uhttp`'s connection closing (or just letting client know about end of file?). Sometimes it hits the `menu` resource, another time `fs.js`.

rmilecki, 2020-09-30:
Jo pointed out that above commit is quite old while LuCI was working fine few weeks ago. So there had to be some interference of at least two changes.

I keep bisecting and I noticed that `uhttpd` regression (described above) was exposed by my LuCI commit bc6e4dfe2916 ("luci-base: use actual JSON-RPC for verifying ubus RPC URL"). It cannot be simply reverted as it will break LuCI due to the `uhttpd` change adding new API.

olimpiumarius, 2020-10-03:
[image]
[image]

closed 2020-10-04 by jow-