openwrt/luci #4469: DHCP limit can be set over safe value, causing router to not boot up

okias, 2020-09-24:
Luci allow setting DHCP out-of-range, which leads to state, where DHCP won't start next boot. (since `100+250` > `255`).

For many routers, there is definition of DHCP range by start and end address, which can be easily overlooked / mistaken on luci.

I propose to check with some easy math JS, which will restrict start + limit < 255, otherwise changes field color to red.
[image]

jow-, 2020-09-24:
Works fine for me, `ipcalc.sh`, which is used by the dnsmasq init script, is capping the value to the maximum allowed value within the subnet mask:

```
root@jj:~# ipcalc.sh 10.11.12.13 255.255.255.0 100 250
IP=10.11.12.13
NETMASK=255.255.255.0
BROADCAST=10.11.12.255
NETWORK=10.11.12.0
PREFIX=24
START=10.11.12.100
END=10.11.12.254
```
I also did put `option limit 250` into `/etc/config/dhcp` and restarted dnsmasq which worked fine.
The generated config also contained a sane value:
```
root@jj:~# grep dhcp-range /var/etc/dnsmasq.conf.cfg01411c 
dhcp-range=set:lan,10.11.12.100,10.11.12.254,255.255.255.0,12h
```

okias, 2020-09-24:
Maybe my "not working DHCP" issue was caused by something else, since it was production router, when I changed value back and it booted, I haven't tried again. I'll try investigate more.

closed 2020-10-30 by jow-