openwrt/luci #4470: ksmb app no longer works, impossible to create network shares

borkra, 2020-09-25:
- Form to enter share information not visible
- Hitting save on the form result in "Unexpected reply data format." message

jow-, 2020-09-25:
- Please mention the exact version of LuCI you're running
- Please try commenting out `option ubus_prefix '/ubus'` in `/etc/config/uhttpd`, close and reopen the LuCI browser tab or try logging in using an incognito window and see if the issue persists.

borkra, 2020-09-25:
@jow Version everything master, fetched from git and rebuilt everything yesterday evening.

luci-base - git-20.267.53658-b17ec09
uhttpd - 2020-09-18-47c34bd6-1

Commenting out "option ubus_prefix '/ubus'" fixes ksmbd

One more issue constant hangs after login and few other pages. If wait long enough get "XHR request timed out"
Tried with browser Firefox 81 and Edge/Chrome 85, on Linux and Windows.

With commented "option ubus_prefix '/ubus'", get "No related RPC reply" message on Log In and Log Out, and maybe other screens

borkra, 2020-10-04:
It looks like this commit causing the issue:
https://git.openwrt.org/?p=project/uhttpd.git;a=commit;h=11723570af9cb7bd87842e79c85ee99530be9902

Reverting uhttpd to the 2020-08-05 version fixes the problem.

borkra, 2020-10-04:
@rmilecki could you please look into this uhttpd issue

closed 2020-10-04 by jow-