openwrt/luci #4473: luci-proto-bonding is not compatible with bonding.sh

qiuzi, 2020-09-26:
option slaves'eth2' can be correctly identified

	list slaves'eth2' is not recognized

he-ma, 2020-09-28:
@qiuzi 
Your problem seems to be the same or similar as I encountered some time ago and I opened an issue for this (https://github.com/openwrt/packages/issues/11455). But as you can see there I made a workaround which should work.

Do you have some more information for me what happens at yours (in the LuCi or during execution of the shell script, ...), which version do you use, your current branch, ...

Regards

qiuzi, 2020-09-28:
@he-ma 
My method is to change "form.MultiValuev" to "form.Value"

//o.validate = updatePrimaries;

he-ma, 2020-09-28:
Unfortunately, I cannot follow you.
For what reason did you alter the code?
What are you going to do?
Can you give me an example?
Have you read my my issue?
What exactly is not working?

qiuzi, 2020-09-28:
@he-ma At this stage, no case can support the variables of the "list" table
Only manually write "eth1 eth2"
So it is more appropriate to change the source code of luci
[https://github.com/openwrt/luci/blob/master/protocols/luci-proto-bonding/htdocs/luci-static/resources/protocol/bonding.js#L177](url)

closed 2021-04-13 by feckert