openwrt/luci #4475: WG intercace public key is not created, and the one of peer is displayed.

nerdoc, 2020-09-27:
openwrt-19.07 branch (git-20.247.75781-0d0ab01) / OpenWrt 19.07.4 r11208-ce6496d796 

I created a new interface named `wg`, entered a private key, and filled out the basic needed forms, and clicked on save, even rebooted the router. 
[image]

I did not yet add a peer, as this is not needed by now - I want to add an android device by displaying the QR code.
But at the wireguard status page, no QR code is displayed, and it tells me that "there is no public key available" for this interface.

IMHO this is clearly wrong, as the pubkey should be calculated from the private key.

When I add a peer (any IP, test pubkey 9YzLmC73/vH5sHTFDEXK3RXknM8RUrqs51F2/sui4Gc=) and click on save + reboot, the QR code appears and displays the pubkey of the peer:
[image]

BUT: this is not the pubkey of the interface itself, but the one of the peer it should connect to.

Even if I delete the peer agian, this pubkey stays in the wireguard status.

----------------------
What I expect:
1. add a wireguard device in Luci, let Luci (by calling `wg privatekey`) create a new private key. Even without adding a peer, the pubkey should be created and displayed readonly.
2. when displaying the QR code (which I can scan from e.g. the android app), display the public key as well in cleartext, or even better: add a button next to it to save this newly created peer as peer in the wireguard config under `/luci/admin/network/network -> <wg interface>`. So I can scan and setup this peer on my phone, and with a few clicks, on the router.

But the current behaviour is definitely wrong (the displayed pubkey is not the one of the wg interface) - and could be done much easier.

nerdoc, 2020-09-27:
This is funny, after deleting peer a few times and rebooting, it works as expected. Maybe this is just the case after first time setup?

nerdoc, 2020-09-27:
Can anyone confirm this? If not, please close it, and I'll file a separate bug for the "one-click-experience" enhancement to create a peer.

castillofrancodamian, 2020-09-27:
It seems to me that it is incorrect that the public key is generated from the private one.
https://forum.openwrt.org/t/duckdns-openwrt-on-wireguard/65354/30?u=castillofrancodamian

n-thumann, 2020-10-05:
I tried to reproduce on `master` @ `LuCI Master (git-20.279.67391-61a4fca) / OpenWrt SNAPSHOT r14576-422cb80ecd` and everything works as expected. I created a new WireGuard interface, let LuCI generate the private key and save. Then I was immediately able to see the config and the QR code on the status page.
I guess that the [output](https://github.com/openwrt/luci/blob/master/applications/luci-app-wireguard/luasrc/view/wireguard.htm#L22) of `wg show` is (for whatever reason) not as expected and therefore LuCI mistakenly parses the clients pubkey as the the server ones.
Regarding the one-click-experience: This definitely could be done better, e.g. include the WireGuard Interface IP as an Endpoint in the QR code and like you suggested saving the QR code config as a peer 👍

nerdoc, 2020-10-07:
Ok. I'll test that today in the evening. IMHO LuCI does not create a PK, I had to create it myself and enter it in the corresponding field.
Regarding the UI workflow for config saving, should I create a new issue to keep discussions separated?

nerdoc, 2020-10-07:
Definitely. When creating a Wireguard interface, no PrivateKey is created by LuCI.
[image]
Maybe I have to upgrade LuCI (which is not recommended, better wait for the whole OpenWRT upgrade image, right?)

johnnyslee, 2020-10-14:
The WireGuard Status page is showing the server public key correctly, but it's confusing us for not indicating it's the *Server* public key.
Another problem is how we are supposed to send the client/peer public key back from the Android app. (We're just too lazy to actually type it.) It would be nice to have a text-box, containing the client/peer public key, near the qr-code for us to copy it.

nickberry17, 2020-12-27:
>But at the wireguard status page, no QR code is displayed
To fix this you need to install `qrencode` e.g.: `opkg update && opkg install qrencode`

nerdoc, 2020-12-27:
@nickberry17 No, sorry, this is not the fix, qrencode *is* installed. I suppose you haven't read the whole thread...?