openwrt/luci #4479: luci-app-samba4: Unexpected reply data format

kongfl888, 2020-09-29:
Look for help.

Just trying to modify an option

[image]

the config still is the initial.
[image]

kongfl888, 2020-09-29:
/$ opkg list-installed luci-base
luci-base - git-20.267.56601-1da9df8
/$ opkg list-installed uhttpd
uhttpd - 2020-09-18-47c34bd6-1

try commenting out option ubus_prefix '/ubus' in /etc/config/uhttpd, restart uhttpd, close and reopen the LuCI browser tab .

it work...
why?
@jow- 

but I have the same problem "XHR request timed out" #4470

dibdot, 2020-09-29:
I can confirm this bug on latest snapshot and other input areas which are using fs.write functions like "Scheduled tasks" ...
[image]

jow-, 2020-09-29:
It is most likely an uhttpd regression on the OpenWrt side.

wevsty, 2020-10-01:
luci-app-adblock page blacklist 
The same problem can be found when saving.

borkra, 2020-10-04:
It looks like this commit causing the issue:
https://git.openwrt.org/?p=project/uhttpd.git;a=commit;h=11723570af9cb7bd87842e79c85ee99530be9902

Reverting uhttpd to the 2020-08-05 version fixes the problem.

closed 2020-10-04 by jow-

kongfl888, 2020-10-05:
It works, thank you!