openwrt/luci #4497: bug in acl.d permission?

Ansuel, 2020-10-06:
In converting the advanced-reboot application I notice that the fs.stat doesn't follow acl rules.
I don't have any rules for '/usr/sbin/fw_printenv' but still fs.stat correctly gives me output.
@jow- is this intended or a bug? Also it would be good to also provide the fs.access api to reduced the data even more.

jow-, 2020-10-06:
@Ansuel - `luci-base` grants filesystem-wide stat permissions yet, mainly for compatibility with the the CBI filebrowser widget. See https://github.com/openwrt/luci/blob/master/modules/luci-base/root/usr/share/rpcd/acl.d/luci-base.json#L14 (the `fs.stat` ubus call uses the `list` permission).

So in light of this, the behavior you see is expected.

Ansuel, 2020-10-06:
Should I set the permission in the acl for the list or I can skip it?

jow-, 2020-10-06:
I think it is good practice to add the required permission anyway to your ACL spec in case luci-base ever revises its default permissions.

closed 2020-10-06 by Ansuel